ZaneL/Teensy-ICM-20948 #2: How to get derived sensor values (rotation vector, game rotation vector, gravity, linear acceleration) from the DMP?

n2d7, 2020-11-25:
Thanks for the great library. Can you give some advice on how to get the derived sensor values such as linear acceleration from the sensor ?

ZaneL, 2020-11-25:
For linear acceleration look in [Icm20948Setup.c](https://github.com/ZaneL/Teensy-ICM-20948/blob/master/src/Icm20948Setup.c)

```
if (inv_icm20948_ctrl_androidSensor_enabled(s, ANDROID_SENSOR_LINEAR_ACCELERATION) && !skip_sensor(s, ANDROID_SENSOR_LINEAR_ACCELERATION)) {
    float linacc_float[3];
    long linAccQ16[3];
    long accelQ16[3];

    /* Compute linear acceleration data based on accelerometer data in Q16 g and on gravity data in Q16 g */
    accelQ16[0] = (int32_t)((float)(accel_float[0]) * (1ULL << 16) + ((accel_float[0] >= 0) - 0.5f));
    accelQ16[1] = (int32_t)((float)(accel_float[1]) * (1ULL << 16) + ((accel_float[1] >= 0) - 0.5f));
    accelQ16[2] = (int32_t)((float)(accel_float[2]) * (1ULL << 16) + ((accel_float[2] >= 0) - 0.5f));

    inv_icm20948_augmented_sensors_get_linearacceleration(linAccQ16, gravityQ16, accelQ16);
    linacc_float[0] = INVN_FXP_TO_FLT(linAccQ16[0], 16);
    linacc_float[1] = INVN_FXP_TO_FLT(linAccQ16[1], 16);
    linacc_float[2] = INVN_FXP_TO_FLT(linAccQ16[2], 16);
    s->timestamp[INV_ICM20948_SENSOR_LINEAR_ACCELERATION] += s->sensorlist[INV_ICM20948_SENSOR_LINEAR_ACCELERATION].odr_applied_us;
    handler(context, INV_ICM20948_SENSOR_LINEAR_ACCELERATION, s->timestamp[INV_ICM20948_SENSOR_LINEAR_ACCELERATION], linacc_float, &accel_accuracy);
}
```

You can see that they get the linear acceleration for x,y,z...linacc_float[0] is x, linacc_float[1] is y, linacc_float[2] is z. Then some function gets called with linacc_float as an argument. 

I think that function is build_sensor_event_data(), which is here:
https://github.com/ZaneL/Teensy-ICM-20948/blob/master/src/Teensy-ICM-20948.cpp

It's missing a section to memcpy the linear acceleation over because [SensorTypes.h](https://github.com/ZaneL/Teensy-ICM-20948/blob/master/src/SensorTypes.h) is missing a linear acceleation field in the inv_sensor_event_t. As far as I can tell invensense never bothered fully implementing the linear acceleration feature in their code.

Anyway, I would try adding a linear acceleation field into that struct, add a memcpy in the build_sensor_event_data() and then add a couple lines in TeensyICM20948::init()

Maybe something like this...
rc = inv_icm20948_set_sensor_period(&icm_device, idd_sensortype_conversion(INV_SENSOR_TYPE_LINEAR_ACCELERATION), 1000 / settings.linear_acceleration_frequency);
rc = inv_icm20948_enable_sensor(&icm_device, idd_sensortype_conversion(INV_SENSOR_TYPE_LINEAR_ACCELERATION), settings.enable_linear_acceleration);

If I have extra time in the next few weeks I'll see if I can get it to work.

n2d7, 2020-11-25:
Thank you very much for the detailed explanation. I will try it out. Really appreciate the help.

n2d7, 2020-11-25:
I implemented it and it works fine. Thanks again for the clear directions.

ZaneL, 2020-11-25:
Np - glad to hear that it works!

closed 2020-12-03 by n2d7

Glavin001, 2021-04-07:
@n2d7 : Would you be interested in sharing your final implementation which worked for you? I'm sure others would benefit -- I know I would, trying to accomplish the same with ICM-20948. Thanks!

isouriadakis, 2021-05-08:
Hi, Thanks for this amazing library.
 I modified the library to include SPI and i2c communication along with linear acceleration, gravity vector, activity recognition, and step counter. 

 You can find it here: https://github.com/isouriadakis/Arduino_ICM20948_DMP_Full-Function
 
I hope it helps someone.