ZaneL/Teensy-ICM-20948 #8: Output Data Rate

FirstIMU, 2021-12-10:
The comment say the output rate is 1 Hz by default. However, I seem to get like 200Hz if I only enable accelerometer. It is 1 Hz only if I also enable the gyroscope as well. Is this expected behaviour ?

ZaneL, 2021-12-10:
I don't have time to check this, but that's not the expected behavior. I've honestly never tested it with everything turned off but the accelerometer output.

For now I guess you can call **icm20948.task()** every one second, rather than as often as possible, and that would give you 1 Hz accel output.

FirstIMU, 2021-12-13:
I am more interested in trying to understand how to setup the chip correctly so the output is meaningful. My initial goal is to understand how to set the rate the DMP is running, the output data rate and  the lpf rate for those sensors. 
It just surprised me that the output rate accel depends on the enabling of the gyro. Enabling the gyro seem to allow the output rate of accel to be controlled according to the formula. I can always ignored the gyro output.

joshgalvan, 2023-08-22:
I understand this is an old post, but @ZaneL would you be able to point me in the direction you found the ODR of the DMP? And where you found the max ODRs of the accel, gyro, and mag?

Reading the datasheet it says the sample rate of the gyro should be up to 9000 with no DLPF (although I would be using a DLPF), along with the other sensors being different too.

I'm trying to figure out certain timings for a project I'm working on and am curious about what rate the DMP is giving me a quaternion at and at what rate the underlying data it's using to calculate said quaternion is changing (the 9 axes).

If you could help, that would be great!

Thank you.