ZaneL/Teensy-ICM-20948 #9: Is your code too big?

nsmith1024, 2021-12-27:
I was thinking of using your code, but it seems your code plus the 20948 library plus my code is too big for the arduino.

All i need is to be able to read the values as a quaternion, do i need all your code to do that?

ZaneL, 2021-12-27:
Yeah, I've only tested this library on Arduino platforms with a large amount of flash and RAM - such as a Teensy.

If I were you I'd first try this library that uses the internal DMP hardware: https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary

If that's still too large, you're gonna have to get the raw accelerometer, mag, and gyro values using a library like that one. But rather than using the DMP to get the quaternion, you'll need to do it using a madgwick algorithm.

https://github.com/arduino-libraries/MadgwickAHRS

The code size should be smaller, but it will require the CPU on your Arduino to do much more work.

ZaneL, 2021-12-27:
https://github.com/arduino-libraries/MadgwickAHRS/blob/master/src/MadgwickAHRS.h

`void update(float gx, float gy, float gz, float ax, float ay, float az, float mx, float my, float mz);`

For example you can see here what you need to do. Get the raw accelerometer, gyro and mag values and feed them into that update function. Then you can get the roll, pitch and yaw.